ARM-software/devlib #20: pexpect error after reboot

jlelli, 2016-02-23:
After the LinuxTarget object is recreated after a reboot we get this error after several commands are executed. Not sure how to give more information, but maybe the error output below is already informative per-se? :-)

```bash
04:04:37  DEBUG   : sudo -- sh -c 'echo pippo'
Failure: EOF (End Of File (EOF). Exception style platform.
<pexpect.pxssh.pxssh object at 0x4af25d0>
command: /usr/bin/ssh
args: [u'/usr/bin/ssh', u'-q', u'-l', u'root', u'10.1.208.164']
searcher: None
buffer (last 100 chars): ''
before (last 100 chars): '\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 255
flag_eof: True
pid: 25597
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1) ... ERROR

======================================================================
ERROR: Failure: EOF (End Of File (EOF). Exception style platform.
<pexpect.pxssh.pxssh object at 0x4af25d0>
command: /usr/bin/ssh
args: [u'/usr/bin/ssh', u'-q', u'-l', u'root', u'10.1.208.164']
searcher: None
buffer (last 100 chars): ''
before (last 100 chars): '\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 255
flag_eof: True
pid: 25597
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/nose/loader.py", line 519, in makeTest
    return self._makeTest(obj, parent)
  File "/usr/local/lib/python2.7/dist-packages/nose/loader.py", line 566, in _makeTest
    return self.loadTestsFromTestCase(obj)
  File "/usr/local/lib/python2.7/dist-packages/nose/loader.py", line 490, in loadTestsFromTestCase
    return super(TestLoader, self).loadTestsFromTestCase(testCaseClass)
  File "/usr/lib/python2.7/unittest/loader.py", line 56, in loadTestsFromTestCase
    loaded_suite = self.suiteClass(map(testCaseClass, testCaseNames))
  File "/media/hdd1tb/work/lisa/tests/eas/rfc.py", line 30, in __init__
    super(RFC, self).__init__(TESTS_CONF, *args, **kwargs)
  File "/home/juri/work/lisa/libs/utils/test.py", line 55, in __init__
    self._runExperiments()
  File "/home/juri/work/lisa/libs/utils/test.py", line 73, in _runExperiments
    self.executor.run()
  File "/home/juri/work/lisa/libs/utils/executor.py", line 136, in run
    if not self._target_configure(tc):
  File "/home/juri/work/lisa/libs/utils/executor.py", line 290, in _target_configure
    self._setup_kernel(tc)
  File "/home/juri/work/lisa/libs/utils/executor.py", line 178, in _setup_kernel
    self._setup_rootfs(tc)
  File "/home/juri/work/lisa/libs/utils/executor.py", line 198, in _setup_rootfs
    , as_root=True)
  File "/home/juri/work/lisa/libs/devlib/devlib/target.py", line 251, in execute
    return self.conn.execute(command, timeout, check_exit_code, as_root)
  File "/home/juri/work/lisa/libs/devlib/devlib/utils/ssh.py", line 155, in execute
    output = self._execute_and_wait_for_prompt(command, timeout, as_root, strip_colors)
  File "/home/juri/work/lisa/libs/devlib/devlib/utils/ssh.py", line 204, in _execute_and_wait_for_prompt
    timed_out = self._wait_for_prompt(timeout)
  File "/home/juri/work/lisa/libs/devlib/devlib/utils/ssh.py", line 222, in _wait_for_prompt
    while not self.conn.prompt(1):
  File "/usr/local/lib/python2.7/dist-packages/pexpect/pxssh.py", line 375, in prompt
    i = self.expect([self.PROMPT, TIMEOUT], timeout=timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 315, in expect
    timeout, searchwindowsize, async)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 339, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 102, in expect_loop
    return self.eof(e)
  File "/usr/local/lib/python2.7/dist-packages/pexpect/expect.py", line 49, in eof
    raise EOF(msg)
EOF: End Of File (EOF). Exception style platform.
<pexpect.pxssh.pxssh object at 0x4af25d0>
command: /usr/bin/ssh
args: [u'/usr/bin/ssh', u'-q', u'-l', u'root', u'10.1.208.164']
searcher: None
buffer (last 100 chars): ''
before (last 100 chars): '\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 255
flag_eof: True
pid: 25597
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1

----------------------------------------------------------------------
Ran 1 test in 0.002s

FAILED (errors=1)
```

setrofim, 2016-02-25:
I just pushed a fix for this. Please let me know if you're still having this issue after updating.

jlelli, 2016-03-04:
Issue is gone. Thanks a lot!

closed 2016-03-04 by jlelli