ARM-software/devlib #26: Android target: su command required

mdigiorgio, 2016-03-07:
When the target is an Android device the `su` command is needed on the target.

In my case, I pushed the `su` binary from the AOSP repository I have in my work environment. It would be necessary to either tell the user he needs to do something similar or provide that binary with devlib and deploy it on the target right after connecting to it (as it is done with `busybox` for example).

setrofim, 2016-03-07:
Please could you describe the issue in more detail? Which part is failing? What device are you running on? For rooted devices and dev boards, `su` should already be in the file system. For unrooted devices, it should be be asked for (unless you explicitly try to use functionality that requires root).

mdigiorgio, 2016-03-07:
I'm running experiments on a N5X. The device is rooted and `su` was not in the filesystem at the beginning, so I got some errors during initialization when trying to execute commands with `as_root=True` .

Hence, I had to pull the binary in the device to make it work.

Now, since in my case executing commands as root is not necessary because I already get a root shell, I thought we could just ignore the `as_root` parameter (in such cases). For doing that, we could change the `execute()` method in the following way:

```python
def execute(self, command, timeout=None, check_exit_code=True, as_root=False):
        if  self.connected_as_root:
            as_root = False
        return self.conn.execute(command, timeout, check_exit_code, as_root)
```

If you want I can send a PR with this modification. What do you think?

douglas-raillard-arm, 2023-12-18:
It might be a good idea to use `busybox su` but at the same time this means `busybox` needs to be suid, which in turn would pre-require root on the target to achieve. Since we haven't experienced that issue in recent years, maybe we should close that @marcbonnici ?