ARM-software/devlib #32: Failed to set CPUFreq ondemand governor's sampling_rate for Linux kernel 4.4

Leo-Yan, 2016-04-26:
Hi,

I saw Lisa set ondemand governor's sampling_rate in the folder: /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate; but when I run Linux kernel 4.4 the folder has been changed as: /sys/devices/system/cpu/cpufreq/policy0/ondemand/sampling_rate and /sys/devices/system/cpu/cpufreq/policy1/ondemand/sampling_rate.

So finally this will introduce failure when run LISA for Juno's profiling with kernel 4.4.

Thanks,
Leo Yan

derkling, 2016-04-27:
We should probably make the hard-coded path in:
  https://github.com/ARM-software/devlib/blob/master/devlib/module/cpufreq.py#L45
more generic. Which means, for example, at __init__ time to probe which path as to be used.

derkling, 2016-05-12:
@setrofim any update on this side? Maybe WA implementation already address this issue?
The "policyN" format seems to be standard for kernel >v4.4 with more than one frequency domain.