ARM-software/devlib #55: target.kick_off doesn't do what it says in the tin on Android targets

JaviMerino, 2016-08-26:
`target.kick_off()` advertises itself as "Like execute but closes adb session and returns immediately, leaving the command running on the device".  However, in `AndroidTarget` the implementation just effectively runs `nohup command`, which doesn't execute it in the background:
```
target.kick_off("sleep 5")

2016-08-26 13:57:36,762 DEBUG   : adb -s 10.1.210.61:5555 shell "cd /data/local/tmp/devlib-target && /data/local/tmp/bin/busybox nohup sleep 5"
2016-08-26 13:57:37,768 DEBUG   : 7919 timed out; sending SIGKILL
```

We have a similar issue with rt-app.  `target.kick_off("rt-app my_task.json")` spawns an rt-app but then kills it after 1 second.  From the documentation, you would expect that rt-app would continue running on the device after `kick_off()` returns.

setrofim, 2016-08-26:
`nohup` should prevent `rt-app` from being killed when the adb session is... we will look into this.

setrofim, 2016-08-26:
I have tried reproducing the issue  by running the following code on an unrooted Note3:

```
import os
import time
from devlib import AndroidTarget

t = AndroidTarget()
print "installing dhrystone"
dhry = t.install(os.path.abspath('dhrystone'))
print "kicking off dhrystone"
t.kick_off('{} -t 1 -r 10000'.format(dhry))
print "waiting..."
time.sleep(5)
pids = t.get_pids_of('dhrystone')
if pids:
    print "Dhrystone PID(s): {}".format(', '.join(map(str, pids)))
    t.killall('dhrystone')
else:
    print "No dhrystone found... :("
```

And  got the following output:

```
$python test.py 
installing dhrystone
kicking off dhrystone
waiting...
Dhrystone PID(s): 5484
```
So `kick_off` appears to be working as expected... Please could you provide more detail about how you're executing that command.

JaviMerino, 2016-08-26:
Tried it on my Juno with Android LSK 16.06, got "No Dhrystone found".  This is what I used:

``` python
import os
import time
import devlib

t = devlib.AndroidTarget(platform=devlib.platform.arm.Juno(),
                              connection_settings={"device": "10.1.210.61:5555"})
print "installing dhrystone"
dhry = t.install(os.path.abspath('dhrystone'))
print "kicking off dhrystone"
t.kick_off('{} -t 1 -r 10000'.format(dhry))
print "waiting..."
time.sleep(5)
pids = t.get_pids_of('dhrystone')
if pids:
    print "Dhrystone PID(s): {}".format(', '.join(map(str, pids)))
    t.killall('dhrystone')
else:
    print "No dhrystone found... :("
```

Output:

```
installing dhrystone
kicking off dhrystone
waiting...
No dhrystone found... :(
```

More complicated for rt-app:
``` python
import devlib
import os
import time
import trappy

rtapp_json_fname = "/tmp/busy_threads.json"
with open(rtapp_json_fname, "w") as fout:
    fout.write("""{
    "global": {
        "calibration": 124,
        "default_policy": "SCHED_OTHER",
        "duration": -1,
        "logdir": "/data/local/tmp/devlib-target"
    },
    "tasks": {
        "busy_thread0": {
            "loop": 1,
            "phases": {
                "p000001": {
                    "loop": 1,
                    "run": 9000000
                }
            },
            "policy": "SCHED_OTHER"
        }
    }
}""")

target = devlib.AndroidTarget(platform=devlib.platform.arm.Juno(),
                              connection_settings={"device": "10.1.210.61:5555"},
                              load_default_modules=False)

ftrace = devlib.FtraceCollector(target, events=["sched_switch"], autoreport=False)

devlib_home = "/data/local/tmp/devlib-target"
target.push(rtapp_json_fname, devlib_home)

rt_app = "/data/local/tmp/bin/rt-app"
remote_json = os.path.join(devlib_home, os.path.basename(rtapp_json_fname))

ftrace.start()
out = target.kick_off("{} {}".format(rt_app, remote_json))
time.sleep(9)
ftrace.stop()

trace_file = "/tmp/rt-app_trace.dat"
ftrace.get_trace(trace_file)

trappy.plotter.plot_trace(trace_file)
```

If you see the trace, rt-app dies after one second.  Use `execute()` instead of `kick_off()` and you will see that rt-app runs for 9 seconds.

setrofim, 2016-08-30:
Ok, so it sounds like nohup is not working properly on that android build. Could you confirm by running something with nohup manually and checking whether it remains running after you kill the spawning shell?

derkling, 2016-08-30:
I've tried this example:

```shell
adb shell "nohup /data/local/tmp/bin/rt-app /data/local/tmp/devlib-target/simple_00.json"
```

Donno if it's the way you mean by "kill the spawning shell" but, if I kill the adb command with a CTRL+C that the rt-app instance also is terminated.

Actually, if from an adb shell you run this simple command:
```shell
nohup sleep 3
```
you can see that the command is not returning for 3s... which means that it's not executed in a detached shell.

setrofim, 2016-08-30:
mm, could you try that while backgrounding the process? i.e.

    adb shell "nohup /data/local/tmp/bin/rt-app /data/local/tmp/devlib-target/simple_00.json &"

Followed by CTRL-C'ling out of that command.

derkling, 2016-08-30:
This is working, this is what we noticed with @JaviMerino last week.
We usually always use nohup with an explicit '&' at the end and that's why we was a bit surprised by the current devlib's implementation to the point to open this ticket.

setrofim, 2016-08-30:
I've added the missing '&' in the Android implementation. I've tested it on my local  device and it worked; but it was working before anyway, so that's not proving much. Please could test the latest commit on your setup and confirm that the issue is now resolved?

setrofim, 2016-09-28:
Confirmed resolved by https://github.com/ARM-software/devlib/commit/ca0b6e88a15ca999b02b97f58f9c7720483aab4c

closed 2016-09-28 by setrofim