ARM-software/devlib #67: Can't use `adb_shell` with `check_exit_code` & `as_root` if `'` in command

bjackman, 2016-10-31:
With both of these options, the command is passed through `escape_single_quotes` twice. This function doesn't appear to work in that context, since it replaces instances of `'` with `'\''` which AFAICT is intended to be a closing quote, an escaped quote, and a re-opening quote - this pattern doesn't nest.

Example: if the command is `ls '-l' '/data'` (note that this already contains single quotes):
- First `escape_single_quotes` (for `as_root`) results in 
```
echo 'ls '\''-l'\'' '\''/data'\''' | su
```
- Second `escape_single_quotes` (for `check_output`) results in 
```
adb shell '(echo '\''ls '\''\\'\'''\''-l'\''\\'\'''\'' '\''\\'\'''\''/data'\''\\'\'''\'''\'' | su); echo "
$?"'
```
So the string that ends up being piped into `su` is `ls \-l\ \/data\`

Haven't had time to investigate the issue further. Might be able to fix it by using double quotes & escapes instead (I believe double quotes are more flexible in Posix shells at least).

marcbonnici, 2016-11-02:
This should be fixed with commit b59f7c360e60e6d0f0b19f41dce11aeec511333a. Please could you confirm if this fixes the issue you were having?

bjackman, 2016-11-02:
Can confirm that this fixes AndroidTarget::push with `as_root=True` where the command is `"cp '{}' '{}'".format(...)`. Thanks a lot for fixing this!

closed 2016-11-02 by bjackman