ARM-software/devlib #118: Turning off USB charging of the target from the energy meter

joelagnel, 2017-05-02:
Can we instruct the target to turn off USB charging from the energy meter as a hint?

Currently this has to be done manually either from target.execute or from adb. Can we automate this by providing some per-device abstraction?

CC @bjackman @derkling

bjackman, 2017-05-03:
`AndroidTarget` has a property for this: https://github.com/ARM-software/devlib/blob/master/devlib/target.py#L1092

As far as it's been tested, it currently only works on Pixel, but if another device has a different method for disabling charging then it should be possible to override the mechanism.

joelagnel, 2017-05-03:
Thanks @bjackman for pointing that out.
Is it Ok if we provide a way to override the path from: libs/utils/platforms/pixel.json ?
Also, since this is related to emeter configuration, can we have emeter set this to False during start/reset and set to True after stop?

bjackman, 2017-05-04:
>  Is it Ok if we provide a way to override the path from: libs/utils/platforms/pixel.

Maybe we can do that, but that means adding API surface so it musn't be a hack to solve a problem for a specific platform. Dis/enabling charging is really procedural information rather than data so I'd want to be pretty conservative about designing a JSON interface for it. Also, that configuration isn't *technically* necessary, the problem could be solved in devlib by making the `charging_enabled` property more generic.

> Also, since this is related to emeter usage, can we have emeter set this to False during start/reset and set to True after stop? 

Yeah that sounds sensible, probably libs/utils/energy.py is the place to do that - patches welcoms ;)

joelagnel, 2017-05-07:
Turns out charging disable from target side is unreliable on devices. It seems to confuse the host-side, and if the host drops the USB connection, then there isn't any way to re-enable charging because that requires the USB connection to begin with. I think some USB host controllers do monitor VBUS current.

Considering the nature of doing this which platform-specific and the fact that its unreliable, I think we should just use something like USB passthrough where USB automatically disconnected externally by the emeter. I started a discussion here: https://github.com/ARM-software/devlib/issues/121