ARM-software/devlib #136: Freezing userspace task not fully reliable

qperret, 2017-06-22:
Freezing userspace tasks is currently achieved by setting the state of the appropriate cgroup to `FROZEN`. This is done by calling `target.write_value` which (by default) raises an exception if the read value does not match the written one. If freezing the tasks does not happen promptly, the read value is `FREEZING` instead of `FROZEN` and the following exception is raised:
`TargetError: Could not set the value of /data/local/tmp/devlib-target/cgroups/devlib_cgh3/DEVLIB_FREEZER/freezer.state to "FROZEN" (read "FREEZING")`

The correct behaviour is to probably to re-read in a loop the value after a short delay if `FREEZING` is read until `FROZEN` is reached or to raise an exception in all other cases.

setrofim, 2017-06-23:
I've submitted a pull request with a proposed fix: https://github.com/ARM-software/devlib/pull/137

Please could you review, and let me know if this fixes the issue in your setup.

qperret, 2017-08-29:
This should be fixed by #159.

closed 2017-08-29 by qperret