ARM-software/devlib #225: Possible misbehaviour due to the new default working directory (/sdcard)

derkling, 2017-12-12:
When running on Hikey960, some of our tests get stuck mid-execution, with no apparent reasons.

We can still "adb shell" into the device but we cannot for example reboot the device itself and, as soon as we try to cd into "/sdcard" the device becomes completely stuck.

We dunno what is the reason behind that misbehaviour, however we notice that the same tests always run to completion if we revert 857edbd4 (which has been merged as part of #202).

valschneider, 2017-12-13:
This currently makes our LISA synthetic test suite fail with the HiKey960. @derkling's tests would get stuck when trying execute `mkdir` in sdcard, while I get this failure in my Jenkins tests:

TimeoutError: Timed out: adb -s HIKEY960_0 shell ls -1 /sdcard/devlib-target/run_dir/*task_wmig_0*.log

This could just be a misconfiguration of the board itself and have nothing to do with devlib, but it's a pain nevertheless :(

marcbonnici, 2017-12-13:
I've tried testing some workloads on my Hikey960 and have been unable to reproduce the issue so far.

As a work around I would suggest setting the `working_directory` parameter to use an alternative folder for your device.

jumana-arm, 2017-12-29:
Also broken for Mate9.