ARM-software/devlib #230: Adding CI (e.g. Travis) to devlib

valschneider, 2018-01-16:
Hi,

The issue fixed in PR https://github.com/ARM-software/devlib/pull/229 was raised by LISA's Travis run of https://github.com/ARM-software/lisa/pull/541. Since our Travis setup uses a LocalLinuxTarget to verify some of LISA's functionalities, the same could be done for devlib - after all, a devlib bug was found with this setup.

Is there any plan to add this kind of test to devlib ? I think even something as basic as creating a LocalLinuxTarget, pushing a basic script and executing it would be useful.

derkling, 2018-01-16:
Maybe can be considered kind of overkilling, but a very simple and still working solution for the time being, could be to run the same LISA's Travis setup. Being LISA a devlib client, it already provides some tests which stress devlib APIs... and it will ensure you do not break one of main devlib's users. ;-)

marcbonnici, 2018-01-18:
Hi,
Yes this would definitely be something we would be interested in implementing, and we have added it to our road map.