ARM-software/devlib #254: ImportError: cannot import name ChromeOsTarget

mwasilew, 2018-04-05:
When trying to run Workload Automation from 'next' branch I'm getting the following exception:
```
# wa list instruments
Traceback (most recent call last):
  File "/usr/local/bin/wa", line 16, in <module>
    from wa.framework.entrypoint import main
  File "/usr/local/lib/python2.7/dist-packages/wa/__init__.py", line 12, in <module>
    from wa.framework.output import RunOutput, discover_wa_outputs
  File "/usr/local/lib/python2.7/dist-packages/wa/framework/output.py", line 9, in <module>
    from wa.framework.configuration.execution import CombinedConfig
  File "/usr/local/lib/python2.7/dist-packages/wa/framework/configuration/execution.py", line 7, in <module>
    from wa.framework.configuration.plugin_cache import PluginCache
  File "/usr/local/lib/python2.7/dist-packages/wa/framework/configuration/plugin_cache.py", line 24, in <module>
    from wa.framework.target.descriptor import list_target_descriptions
  File "/usr/local/lib/python2.7/dist-packages/wa/framework/target/descriptor.py", line 5, in <module>
    from devlib import (LinuxTarget, AndroidTarget, LocalLinuxTarget,
ImportError: cannot import name ChromeOsTarget
```
WA was installed on a clean ubuntu 16.04 (docker container). Here is how the package situation looks like:
```
# pip freeze 
certifi==2018.1.18
chardet==3.0.4
colorama==0.3.9
devlib==0.0.4
idna==2.6
Louie==1.1
nose==1.3.7
numpy==1.14.2
pandas==0.22.0
pexpect==4.4.0
ptyprocess==0.5.2
pyserial==3.4
python-dateutil==2.7.2
pytz==2018.3
PyYAML==3.12
requests==2.18.4
six==1.11.0
urllib3==1.22
wa==3.0.0
wrapt==1.10.11
```

The problem doesn't happen when I install devlib from Vincent's fork: https://github.com/vingu-linaro/devlib. Same result with devlib installed from source and from pip package downloaded from pypi.

marcbonnici, 2018-04-05:
Hello,

This is an error that occurs when trying to use a too old version of devlib with WA3. The package on pypi is very much out of date so you will need to install devlib directly from github. If you have already tried installing it from source, is it possible that the old version was also still installed and being used by default? 

Also if you are trying to setup a docker image for use with WA we actually now provide a Dockerfile that will install a known working version of devlib along with WA3 which can be found here: https://github.com/ARM-software/workload-automation/blob/next/extras/Dockerfile

mwasilew, 2018-04-05:
Marco,
You're right. Apparently pip uninstall doesn't clear everything. When installing from sources, it works well. Any idea why the code in pypi is different from the repository tip but they share the same version number? I guess it would be nice to upload new version to pypi or at least bump version in the repository to avoid confusion.

marcbonnici, 2018-04-05:
Ah ok, that's good.
Yea I think the version numbers are a bit of an oversight on our part, the current plan is that for the official release of WA3 we are also going to be releasing devlib version 1.0 and then from then onwards we will be doing more frequent synchronised version releases between the 2, which will also be updated correspondingly on pypi.