ARM-software/devlib #315: devlib indirectly causes kernel panic because of dmidecode

qais-yousef, 2018-08-24:
I've been hitting a weird kernel panic when trying to run Lisa tests on Juno R2 that was root caused to dmidecode.

My userspace had dmidecode automatically installed somehow and it seems that devlib picks up on it and tries to use it to get some info about the hardware/system (guessing), but the tool wrecks havoc on kernel space by trying to mmap() invalid regions, corrupting memory, hence causing a kernel panic.

```
[  372.378964] ---[ end Kernel panic - not syncing: Asynchronous SError Interrupt ]---
[  372.386574] WARNING: CPU: 1 PID: 3813 at kernel/sched/core.c:1882 set_task_cpu+0x2dc/0x348
[  372.386576] Modules linked in:
[  372.386582] CPU: 1 PID: 3813 Comm: dmidecode Not tainted 4.18.0-rc5-g7b6372922599 #25
[  372.386584] Hardware name: ARM Juno development board (r0) (DT)
[  372.386587] pstate: 20000085 (nzCv daIf -PAN -UAO)
[  372.386591] pc : set_task_cpu+0x2dc/0x348
[  372.386594] lr : try_to_wake_up+0x1c4/0x650
[  372.386595] sp : ffff00000800b9f0
[  372.386597] x29: ffff00000800b9f0 x28: 0000000000000002
[  372.386602] x27: ffff000009f5b000 x26: ffff000009f7d000
[  372.386607] x25: ffff8009740e9740 x24: ffff000009f7d000
[  372.386611] x23: 0000000000000080 x22: ffff000009f7d000
[  372.386615] x21: ffff8009740e9f40 x20: 0000000000000002
[  372.386619] x19: ffff8009740e9740 x18: ffffffffffffffff
[  372.386623] x17: 0000000000000001 x16: 0000000000000007
[  372.386627] x15: ffff000009f7d888 x14: ffff00008adc8c1f
[  372.386631] x13: ffff00000adc8c2d x12: 0000000000000000
[  372.386635] x11: 0000000005f5e0ff x10: ffff00000800bbf0
[  372.386639] x9 : 00000167e756a00c x8 : ffff000009f7d000
[  372.386643] x7 : 0000000000000006 x6 : 0000800976017000
[  372.386647] x5 : 0000000000000000 x4 : ffff8009740e9ae0
[  372.386651] x3 : 0000000000000001 x2 : 0000000000000002
[  372.386655] x1 : ffff000009f7db00 x0 : 0000000000000000
[  372.386659] Call trace:
[  372.386662]  set_task_cpu+0x2dc/0x348
[  372.386665]  try_to_wake_up+0x1c4/0x650
[  372.386668]  default_wake_function+0x34/0x48
[  372.386673]  __wake_up_common+0x90/0x158
[  372.386676]  __wake_up_locked+0x40/0x50
[  372.386680]  ep_poll_callback+0xb8/0x378
[  372.386684]  __wake_up_common+0x90/0x158
[  372.386687]  __wake_up_common_lock+0x88/0xd0
[  372.386690]  __wake_up+0x40/0x50
[  372.386695]  wake_up_klogd_work_func+0x4c/0x68
[  372.386701]  irq_work_run_list+0x74/0xc0
[  372.386704]  irq_work_tick+0x48/0x60
[  372.386708]  update_process_times+0x44/0x60
[  372.386712]  tick_sched_handle.isra.5+0x44/0x68
[  372.386714]  tick_sched_timer+0x50/0xa0
[  372.386718]  __hrtimer_run_queues+0x19c/0x6f0
[  372.386722]  hrtimer_interrupt+0xd8/0x250
[  372.386728]  arch_timer_handler_phys+0x38/0x58
[  372.386731]  handle_percpu_devid_irq+0xd0/0x4d0
[  372.386736]  generic_handle_irq+0x34/0x50
[  372.386739]  __handle_domain_irq+0x68/0xc0
[  372.386742]  gic_handle_irq+0x60/0xb0
[  372.386745]  el1_irq+0xb4/0x130
[  372.386749]  panic+0x260/0x2ac
[  372.386754]  __stack_chk_fail+0x0/0x28
[  372.386756]  arm64_serror_panic+0x80/0x90
[  372.386759]  do_serror+0xdc/0xe0
[  372.386762]  el0_error_naked+0x10/0x18
[  372.386764] irq event stamp: 2035
[  372.386770] hardirqs last  enabled at (2035): [<ffff0000082a857c>] handle_mm_fault+0x22c/0x320
[  372.386774] hardirqs last disabled at (2034): [<ffff0000082a842c>] handle_mm_fault+0xdc/0x320
[  372.386779] softirqs last  enabled at (1858): [<ffff0000080879f8>] fpsimd_restore_current_state+0x38/0x78
[  372.386783] softirqs last disabled at (1856): [<ffff0000080879d8>] fpsimd_restore_current_state+0x18/0x78
[  372.386785] ---[ end trace 383dd3590c25ba7c ]--- 
```

In my case removing dmidecode makes things happy again. Which poses the question, is it necessary to use dmidecode given it's not stable/safe?

I think devlib should be more careful about this tool as when speaking to others it seems to have caused problems on other platforms too.

qais-yousef, 2018-08-24:
OK. So you're suggesting to update Lisa to implement this, right?

I don't mind to be honest. But is there no way to get this done transparently within devlib? I'll happily look at contributing the patch if it helps.

setrofim, 2018-08-24:
> OK. So you're suggesting to update Lisa to implement this, right?

My suggestion it to remove `dmidecode` from the device. The workaround above is  in case you have a large number of devices affected. It's up to the LISA team if they want to integrate it into LISA itself, but I would recommend limiting this to a particular deployment, as this is a workaround for a broken setup.

> But is there no way to get this done transparently within devlib? I'll happily look at contributing the patch if it helps.

My understanding is that this is an issue that only affects particular kernels and is limited to a small number of devices (at least, this has been my experience). I'd rather not remove something for all devices in order to deal with a few broken ones.

qais-yousef, 2018-08-24:
OK fair enough. I already did remove dmidecode as soon as I found out the cause. But the fact that I removed it and things just worked is what stirred up this as from my limited point of view it was a NOP. I stress my limited point of view - I am not saying it is a NOP :)

I was on 4.18-rc5 and dmidecode v3. I wouldn't say this was a broken setup - but dmidecode is broken. I wasn't aware of dmidecode existence on the userspace, nor that devlib will try to use it, nor that it can cause the observed panics. It can easily catch the unwary - hence I raised the issue to try to find a safer alternative as from my observed world dmidecode didn't seem to add much.

But what do I know :)

setrofim, 2018-08-24:
> I wouldn't say this was a broken setup - but dmidecode is broken. 

Yes, sorry, perhaps "platform" would have been a more appropriate term than "setup". 

> It can easily catch the unwary - hence I raised the issue to try to find a safer alternative as from my observed world dmidecode didn't seem to add much.

Yes, I appreciate what you're saying. I have looked for an alternative to using `dmidecode` and was not able to find one that is in any way generic (aside from on Android).  If you're aware of a better/safer alternative, please let us know and we'll swap to using that.

And you're right, at the moment we're not using it for much (though that might change in the future -- on platforms where DMI is implemented, `dmidecode` reports a lot of useful information about the hardware); however, this issue also only affects a small number of platforms and is easy to work around (uninstall a tool that is broken anyway). If I'm misjudging the extent to which this issue is prevalent, it may be worth revisiting the use of `dmidecode`...

closed 2018-08-24 by setrofim