ARM-software/devlib #317: target: working_directory issue(s)

v0lker, 2018-08-29:
hi,

i stumbled upon this when i was trying to connect to a linux machine with a user that has an unusual home directory path (`/foo/user/`). not actually a problem as it is easy enough to work around, but it seems wrong:

`LinuxTarget::_resolve_paths()` will set the working directory to `/home/user/devlib-target` and then `Target::connect()` will try to `mkdir -p` that directory.

1. if /home/user doesn't exist, this will try to create it, but it won't succeed for ordinary users, as normally, /home is not writeable for just anybody

this could be solved by adding `as_root=True` to the mkdir invocation. but then, the directory would be owned by root and not any good for the user, so it needs additional fixups (changing ownership of the  directories involved).

2. i think, it should use the users home - which could potentially be retrieved by executing `pwd` or awking the home out of /etc/passwd - and not try to recursively create the devlib-target sub directory.

i'm happy enough to provide a PR.

setrofim, 2018-08-30:
You can override working directory on target creation:

    target = LinuxTarget(working_directory='/foo/user/devlib-working')

However, extracting it from `/etc/passwd` is not a bad idea; please feel free to submit the PR.

marcbonnici, 2018-09-21:
Resolved with https://github.com/ARM-software/devlib/pull/319.

closed 2018-09-21 by marcbonnici