ARM-software/devlib #321: utils/types: Potential naming conflict?

valschneider, 2018-09-20:
Hi,

I was casually opening some file in utils (`version.py`), and my syntax checker exploded:

```python
/usr/bin/pylint3 -r n --output-format text --msg-template \{path\}\:\{line\}\:\{column\}\:\{C\}\:\{symbol\}\:\{msg\} /home/valsch01/Work/lisa/external/devlib/devlib/utils/version.py
Error processing line 1 of /home/valsch01/.local/lib/python3.6/site-packages/sphinxcontrib_websupport-1.1.0-py3.5-nspkg.pth:

Failed to import the site module
Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 174, in addpackage
    exec(line)
  File "<string>", line 1, in <module>
  File "/home/valsch01/Work/lisa/external/devlib/devlib/utils/types.py", line 29, in <module>
    import re
  File "/usr/lib/python3.6/re.py", line 122, in <module>
    import enum
  File "/usr/lib/python3.6/enum.py", line 2, in <module>
    from types import MappingProxyType, DynamicClassAttribute
ImportError: cannot import name 'MappingProxyType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 561, in <module>
    main()
  File "/usr/lib/python3.6/site.py", line 547, in main
    known_paths = addusersitepackages(known_paths)
  File "/usr/lib/python3.6/site.py", line 291, in addusersitepackages
    addsitedir(user_site, known_paths)
  File "/usr/lib/python3.6/site.py", line 213, in addsitedir
    addpackage(sitedir, name, known_paths)
  File "/usr/lib/python3.6/site.py", line 184, in addpackage
    import traceback
  File "/usr/lib/python3.6/traceback.py", line 5, in <module>
    import linecache
  File "/usr/lib/python3.6/linecache.py", line 8, in <module>
    import functools
  File "/usr/lib/python3.6/functools.py", line 22, in <module>
    from types import MappingProxyType
  File "/home/valsch01/Work/lisa/external/devlib/devlib/utils/types.py", line 29, in <module>
    import re
  File "/usr/lib/python3.6/re.py", line 122, in <module>
    import enum
  File "/usr/lib/python3.6/enum.py", line 2, in <module>
    from types import MappingProxyType, DynamicClassAttribute
ImportError: cannot import name 'MappingProxyType'
```

Looking at hints on the net, I stumbled on this: https://stackoverflow.com/questions/43453414/cannot-import-name-mappingproxytype-error-after-importing-functools
And indeed, renaming types.py to something else makes pylint happy.

Is that something related to my setup? I find it strange that such an error didn't show up earlier, considering that `target.py` does `from devlib.utils.types`

setrofim, 2018-09-21:
Were you running pylint with `devlib/futils/` as your `$PWD`? The issue is that local directory imports take precedence over global imports, so if there is a `types.py` or similar in the current working directory, `import types` will import that over the library module.

This is normally not a problem as our `types` is namespaced under `devlib.utils`.

valschneider, 2018-09-26:
Hm so it seems like Emacs runs pylint in whatever directory the file resides in. So yeah, when I open `devlib/utils/*.py`, pylint gets run in `devlib/utils` which causes that issue.

closed 2018-10-18 by valschneider