ARM-software/devlib #329: sudo malfunction not handled gracefully

v0lker, 2018-10-25:
on my root fs, sudo wasn't setuid root
(i guess the output of utils/ssh.py: SshConnection::_execute_and_wait_for_prompt() is probably just one line (with the error message), missing the return value, and i think that's the reason why i'm getting a one-page backtrace stating that the tuple to be split didn't contain enough members (` output, exit_code_text, _ = raw_output.rsplit('\r\n', 2)`).

i propose there should be a check at start of day and it should complain. something like
```
class SshConnection(object):
[...]
    def __init__(self,
[...]
        self.conn = ssh_get_shell(host, username, password, self.keyfile, port, timeout, False, None)                                                                
        atexit.register(self.close)
        try Exception as e:
            output = self.execute("echo hello" ,...)
            if output != "hello":
                raise TargetStableError("executing commands on the target does not seem to work {}".format(e))
[...]
```

i could send a PR if you like...

setrofim, 2018-10-25:
This sounds more like a device-specific hack than a fix. Would it make more sense for this to be in a target-specific adapter, rather than upstream?

v0lker, 2018-10-25:
no, not at all. i must have worded my proposal badly, let me try again.

current behaviour:
devlib throws a longish chain of exceptions with the last one being that a tuple doesn't have enough members.

problem: it is not obvious that it fails due to a misconfiguration of ssh/sudo/bash/...

proposed improvement: 'fail early, in an obvious manner'.
during establishment of ssh connection, it is checked whether commands can actually be executed as that user and an obvious error message is thrown (something like "you cannot execute commands via ssh, please check").

so, it is general and it would belong into `SshConnection`, the question is just whether it's useful or not - if you think that it isn't then that's fine by me.

setrofim, 2018-10-25:
I don't like the solution of executing an arbitrary echo on each and every connection creation (there may be multiple of these per session, as every thread will have its own connection, and the user can always create additional ones). Does this need to be tested for explicitly, or can this just be handled by examining the output in the exception handler on error?

v0lker, 2018-10-25:
good point - yes, certainly. i was just wondering whether you want this in the first place.

so, how about something along these lines:
```
class SshConnection[...]
    def execute[...]
        try:
            output, exit_code_text, _ = raw_output.rsplit (...)
        except ValueError as e:
            raise TargetStableError ("cannot parse ssh command reply - possible misconfiguration on target")
```

setrofim, 2018-10-26:
Yes, something along those lines should work. Though are you certain that this is the only issue that can cause the split to fail? `raw_output` should be examined to see if this is indeed what is going on, and propagated is if the exact problem cannot be identified.

v0lker, 2018-11-07:
it's not the split that fails - it's the tuple assignment.

btw. the adb shell version of this is slightly optimistic:
```
405     if raw_output:
406         try:
407             output, exit_code, _ = raw_output.replace('\r\n', '\n').replace('\r', '\n').rsplit('\n', 2)
408         except ValueError:
409             exit_code, _ = raw_output.replace('\r\n', '\n').replace('\r', '\n').rsplit('\n', 1)
410             output = ''
411     else:  # raw_output is empty
412         exit_code = '969696'  # just because
413         output = ''
```

how about https://github.com/ARM-software/devlib/pull/337 ?

v0lker, 2018-11-19:
i'm happy with #337, thus closing the issue.

as mentioned above, it looks like the adb version could be improved similarly, but if no one sees any issues, it's probably fine the way it is.

closed 2018-11-19 by v0lker