ARM-software/devlib #336: Target.read_tree_values_flat fails on binary files

douglas-raillard-arm, 2018-11-07:
This shell command executed on the target sometimes fail since busybox grep does handle gracefully binary files:
```
/tmp/devlib-target/bin/shutils read_tree_values /sys/class/hwmon 3
```
This results in a non-zero error code, and a TargetStableError is raised at that point. `shutils read_tree_values` should be able to handle binary files, or at least ignore them without a non-zero error code.

marcbonnici, 2018-11-08:
I'm not currently sure of a way to solve this problem reliably so will have to look into this. 
For now if you do not need error checking for the call I would suggest setting `check_exit_code` to `False` when calling `read_tree_values_flat` to prevent an error from being raised.

douglas-raillard-arm, 2018-11-09:
A solution could be to create a tar archive and stream it through base64, then read the file values from Python. All of that can be done in memory using `base64` module to decode the stream into an `io.BytesIO`, and then read it using the `tarfile` module. There is no need to create any temp file, all of that is in the Python standard library and it will work over a UART shell as well (no need for ssh binary pipes). busybox have both `tar` and `base64`.

At some point I'll also need to be able to read files like /proc/device-tree/compatible, which is a null-separated list of strings. Handling this kind of file can only be done if devlib supports binary content somehow.

`check_exit_code` cannot really be used here, since it is in an upstream devlib module (so we don't want hacks there), and I assume it would hide issues like missing files or permissions.

qperret, 2019-01-30:
I guess that was solved by https://github.com/ARM-software/devlib/pull/360 ?

setrofim, 2019-01-30:
I can confirm that the fix works, however, it also seems to introduce a *huge* delay when attempting to read `/sys/devices/system/cpu` on my machine. Could someone else verify this?

If this is confimed, it might be worth adding a flag that allows falling back to the old method.

douglas-raillard-arm, 2019-01-30:
I tested that without observing any slowdown:
```
import devlib
target = devlib.target.LocalLinuxTarget(connection_settings={'unrooted': True})
target.read_tree_values('/sys/devices/system/cpu')

{'isolated': '', 'kernel_max': '319', 'modalias': 'cpu:type:x86,ven0000fam0006mod003C:feature:,0000,0001,0002,0003,0004,0005,0006,0007,0008,0009,000B,000C,000D,000E,000F,0010,0011,0013,0015,0016,0017,0018,0019,001A,001B,001C,001D,001F,002B,0034,003A,003B,003D,0068,006B,006C,006D,006F,0070,0072,0074,0075,0076,0078,0079,007C,0080,0081,0082,0083,0084,0085,0086,0087,0088,0089,008B,008C,008D,008E,008F,0091,0093,0094,0095,0096,0097,0098,0099,009A,009B,009C,009D,009E,00C0,00C5,00E1,00E3,00E7,00EB,00EC,00F0,00F1,00F3,00F5,00F6,00F9,00FA,00FB,00FD,0100,0101,0102,0103,0104,0111,0120,0121,0123,0125,0127,0128,0129,012A,012D,0140,01C0,01C1,01C2,01C4,01C5,01C6,025A,025B,025C,025F', 'offline': '', 'online': '0-7', 'possible': '0-7', 'present': '0-7', 'uevent': ''}
```

Maybe you have a huge binary file that contains a lot of null byte padding or something like that ?

setrofim, 2019-01-30:
Hm maybe, though I did not see an obvious culprit in the output. I tested  on my ``x86_64`` workstation, using ``LocalLinuxTarget``. If the slowdown, is limited to that, we can probably live with  it...

marcbonnici, 2019-01-30:
I tested the above example on my `x86_64` machine and it also returns instantaneously, however if I add the `depth=4` parameter (which is used as part of WA initialization), I see an approximately 8 second delay for the command to return.

douglas-raillard-arm, 2019-01-30:
`depth=4` also slows down the invocation on my setup, which is somehow expected given the number of files that involves:
```
>>> output = target.read_tree_values('/sys/devices/system/cpu', depth=4)
# split on "," is approximate but gives an idea of the number of files+folders
>>> len(str(output).split(','))
25937
```

marcbonnici, 2020-03-20:
Should be addressed by https://github.com/ARM-software/devlib/pull/360 and https://github.com/ARM-software/devlib/pull/366. Please re-open if this is not the case.

closed 2020-03-20 by marcbonnici