ARM-software/devlib #346: SshConnection._scp() displays the password in exception backtrace

douglas-raillard-arm, 2018-11-21:
The `HostError` exception message in `SshConnection._scp()` is built with the "redacted command" that is not supposed to contain the password. However, the exception backtrace of the original `CalledProcessError` with the full password is still displayed by the interpreter using this pattern:

```
<message of CalledProcessError>

During handling of the above exception, another exception occurred:

<message of HostError>
```

douglas-raillard-arm, 2018-11-21:
The relevant part of '_scp()` is:
```
    def _scp(self, source, dest, timeout=30):
       ...
        except subprocess.CalledProcessError as e:
            raise HostError("Failed to copy file with '{}'. Output:\n{}".format(
                command_redacted, e.output))
```

The `HostError` message contains the redacted command, but the `CalledProcessError`'s message contains the full command line, that includes the clear password. It should be noted that the password should be replaced with `str.replace(count=1)` to avoid such situations that obviously reveal the password:
```
sshpass -p <redacted>  scp /my/file <redacted>@myhost.com:/<redacted>/folder 
```

Raising an exception when handling another one will print a backtrace for both by default (at least on Python 3) if that exception bubbles up and is not caught anywhere:
```
try:
    raise ValueError('foo')
except ValueError as e:
    raise IndexError('bar')
```
output:
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
ValueError: foo

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
IndexError: bar
>>> 
```

marcbonnici, 2018-11-21:
Ah ok I see. I have not found a nice way to modify the original backtrack to redact the password from there, however the useful information is replicated in our redacted exception so how about using `raise_from` to just prevent showing the original traceback with something like this https://github.com/ARM-software/devlib/commit/3662c0397fef2bf0172ab1ad500491d803dab644?

douglas-raillard-arm, 2018-11-21:
Looks like it would work. The `str.replace` issue should be solved as a side effect of https://github.com/ARM-software/devlib/pull/339

closed 2019-10-24 by marcbonnici