ARM-software/devlib #394: Connection to target fails

douglas-raillard-arm, 2019-06-11:
Connecting to one Linux target seems to fail for unknown reasons:
  ```
File "/opt/git/tools/lisa/lisa/target.py", line 257, in __init__
    devlib_excluded_modules=devlib_excluded_modules,
  File "/opt/git/tools/lisa/lisa/target.py", line 534, in _init_target
    working_directory = workdir,
  File "/opt/git/tools/lisa/external/devlib/devlib/target.py", line 963, in __init__
    is_container=is_container)
  File "/opt/git/tools/lisa/external/devlib/devlib/target.py", line 258, in __init__
    self.connect()
  File "/opt/git/tools/lisa/external/devlib/devlib/target.py", line 269, in connect
    self.execute('mkdir -p {}'.format(quote(self.working_directory)))
  File "/opt/git/tools/lisa/external/devlib/devlib/target.py", line 393, in execute
    strip_colors=strip_colors, will_succeed=will_succeed)
  File "/opt/git/tools/lisa/external/devlib/devlib/utils/ssh.py", line 217, in execute
    raise TargetStableError(message.format(exit_code, command, output))
devlib.exception.TargetStableError: Got exit code 127
from: PATH=(pwd); __devlib_ec=$?; echo; echo $__devlib_ec
/root/devlib-target/bin:$PATH && mkdir -p '(pwd); __devlib_ec=$?; echo; echo $__devlib_ec
/root/devlib-target'
OUTPUT: (PATH=(pwd); __devlib_ec=$?; echo; echo $__devlib_ec
> 
> /root/devlib-target/bin:$PATH && mkdir -p '(pwd); __devlib_ec=$?; echo; echo $__devlib_ec
> 
> /root/devlib-target'); __devlib_ec=$?; echo; echo $__devlib_ec

0
-bash: /root/devlib-target/bin:pwd: No such file or directory
```

Linux (ssh) target: Ubuntu 14.04
password authentication

It seems that the "$" signs have gone for some reason, but some other parts of the command might also be missing since `mkdir -p $(pwd)` would not make any sense.

setrofim, 2019-06-11:
The line number in the traceback don't seem to match current code. Which version of `devlib` is this? Also would it be possible to confirm whether the issue still occurs with the current tip of `master`?

douglas-raillard-arm, 2019-06-11:
If I'm not mistaken, it's at `8343794d3 module/thermal: Gracefully handle unexpected sysfs names`, which is the current master I see (the lines seems to match with that).

setrofim, 2019-06-11:
Ah, my mistake -- it is I who had the wrong version. My apologies.

douglas-raillard-arm, 2019-10-23:
It's probably tied to a specific setup. Looking at the description (ubuntu 14.04) it was probably a regular machine, and I'm trying to connect to such computer very often so I can't say if the issue is still there or not. 

I guess we can close for now and reopen if we hit that again ...

closed 2019-10-24 by setrofim