ARM-software/devlib #403: SystraceCollector.start() returns before tracing has started

JaviMerino, 2019-07-22:
In `devlib.trace.SystraceCollector`, `start()` returns after executing `subprocess.Popen()` for systrace. That doesn't mean that systrace is running though, the function can return even before systrace has had a chance to execute anything. Therefore that when you run the command you want to trace, systrace will miss the first seconds of the execution. `start()` should return when tracing has started.

The fix is to wait until systrace outputs "Starting tracing (stop with enter)". You would think that something like this would work but it doesn't and I can't figure out why:
``` diff
--- a/devlib/trace/systrace.py
+++ b/devlib/trace/systrace.py
@@ -137,9 +137,12 @@ def start(self):
         self._systrace_process = subprocess.Popen(
             self.systrace_cmd,
             stdin=subprocess.PIPE,
+            stdout=subprocess.PIPE,
+            bufsize=0,
             shell=True,
             universal_newlines=True
         )
+        self._systrace_process.stdout.read(1)
 
     def stop(self):
         if not self._systrace_process:
```

Looks like the stdout for systrace is buffered, but it shouldn't be.

This is not a full reproducer, but a simplified version of the code I'm running is:
``` python
from devlib.trace.systrace import SystraceCollector
from lisa.wlgen.rta import RTA, Periodic

systrace = SystraceCollector(target, categories=['sched'], buffer_size=50 * 1024)
rtapp_profile = {"task": Periodic(duty_cycle_pct=100, cpus=[4], duration_s=1)}
rtapp = RTA.by_profile(target, "task", rtapp_profile)

systrace.start()
rtapp.run()
systrace.stop()

systrace.get_trace("my_trace")
```

If you want to reproduce systrace's behaviour without having to run systrace, you can use this simple program:

``` python
#!/usr/bin/env python2

import time

time.sleep(2)
raw_input('Starting tracing (stop with enter)')
print('Tracing completed. Collecting output...')
```

marcbonnici, 2019-07-25:
I'm unfortunately seeing the same behaviour here, I tried adding a print statement into the script with the same result and only could manage to read the output by also adding an explicit `sys.stdout.flush()`. Therefore I agree with the conclusion that output from the script is being buffered however currently unsure on how to prevent that.

marcbonnici, 2019-07-30:
Looking into this further, if the script is invoked directly with python and we specify the `-u` parameter this should force the output to be unbuffered and we are then able to call the read method as expected. 

e.g.
```
self.systrace_cmd = 'python2 -u {} -o {} -e {}'.format(
            self.systrace_binary,
            self._tmpfile.name,
            self.target.adb_name
        )
```

Would you be able to verify the same behaviour on your end?

JaviMerino, 2019-07-30:
Yes, that fixes it! I'll send a pull request

PR #411 (2019-07-30): trace/systrace: make start() return when tracing has started
In `devlib.trace.SystraceCollector`, `start()` returns after executing
`subprocess.Popen()` for systrace. That doesn't mean that systrace is
running though, the function can return even before systrace has had a
chance to execute anything. Therefore, when you run the command
you want to trace, systrace will miss the first seconds of the
execution.

Run systrace with `-u` to unbuffer its stdin and wait for it to print
"Starting tracing (stop with enter)" before returning.

Fixes #403
commit 71bd8b10 (2019-07-30): trace/systrace: make start() return when tracing has started

In SystraceCollector, start() returns after executing
subprocess.Popen() for systrace. That doesn't mean that systrace is
running though, the function can return even before systrace has had a
chance to execute anything. Therefore, when you run the command
you want to trace, systrace will miss the first seconds of the
execution.

Run systrace with -u to unbuffer its stdin and wait for it to print
"Starting tracing (stop with enter)" before returning.

Fixes #403
closed 2019-07-30 by marcbonnici