ARM-software/devlib #405: commit #387 ('su -c') seems to break command execution for me

v0lker, 2019-07-25:
su doesn't know about the -c option, at least on my test platform with an android q preview.

evil_platform:/ # which su
/system/xbin/su

and that's the only one.. on that machine.

marcbonnici, 2019-07-26:
Would you be able to provide the version of `su` that you are using and also the error output that attempting to use that parameter produces?

v0lker, 2019-07-29:
i'm using a different version but visually, it looks like this one: https://android.googlesource.com/platform/system/extras/+/refs/tags/android-q-preview-5/su/su.cpp

the error message is ```su: invalid uid/gid '-c'```

marcbonnici, 2019-07-30:
A fallback mechanism has been added with https://github.com/ARM-software/devlib/pull/409, would you be able to try this out and see if it resolves your issue?

v0lker, 2019-07-30:
thanks, that seems to work for me:
```
[ins] In [6]: a = devlib.AndroidTarget(working_directory=s)
[...]
[ins] In [8]: a.execute("whoami")         
Out[8]: 'shell\n'

[ins] In [9]: a.execute("whoami", as_root=True)                                     
Out[9]: 'root\n'
```

closed 2019-07-30 by marcbonnici