ARM-software/devlib #407: Failed to start an agenda with cpuidle set for debug

dlezcano, 2019-07-29:
The agenda fails with the error:

2019-07-29 19:57:03,557 ERROR      executor: Got exit code 1
2019-07-29 19:57:03,557 ERROR      executor: from: PATH=/root/devlib-target/bin:$PATH && cat /sys/devices/system/cpu/cpuidle/current_governor_ro
2019-07-29 19:57:03,557 ERROR      executor: OUTPUT: cat: /sys/devices/system/cpu/cpuidle/current_governor_ro: No such file or directory
2019-07-29 19:57:03,557 ERROR      executor:
2019-07-29 19:57:03,560 INFO       executor: Done.

Actually the kernel is compiled with the ability to set the governor from sysfs, thus writing to the file /sys/devices/system/cpu/cpuidle/current_governor file is allowed and this one does no longer end with [_ro].

The devlib should read 'current_governor_ro', if it fails fallback to 'current_governor'

marcbonnici, 2019-07-30:
Would you be able to try out https://github.com/ARM-software/devlib/pull/408 and see if that fixes the problem for you?

setrofim, 2019-10-23:
Assuming this has been resolved as no response from OP.

closed 2019-10-23 by setrofim