ARM-software/devlib #444: dmidecode output unused

douglas-raillard-arm, 2019-11-26:
I've been hitting #315 recently and after looking at the code a little bit, it seems that Platform.model instance attribute is actually never used. Both LinuxTarget and AndroidTarget define their own model property (read from device tree or using adb). If I did not miss anything, that would mean that the model attribute of the platform is dead code, which happens to be harmful in the current situation.

Could that model attribute be either removed or turned into a property in Platform ? This way, dmidecode would only be called when someone looks at platform.model and not by default upon target setup.

setrofim, 2019-11-28:
I believe https://github.com/ARM-software/devlib/pull/447 should address this -- it moves the model code from `LinuxTarget` and `AndroidTarget` into `Platform._set_model_from_target()` and uses that in preference to the `dmidecode` method (which is still used as a fall-back). The `Target` property now just returns the underlying `Platform` attribute (note: it still has to be an attribute in the platform because `Platform` does not have direct access to the target outside of `upadate_from_target()` as it is itself an attribute of `Target`, and we don't want a cycle).

douglas-raillard-arm, 2019-11-28:
That should do the trick as ARM/ARM64 should have that device tree entry, thanks :)

closed 2019-12-27 by douglas-raillard-arm