ARM-software/devlib #468: Local ping to Android device even when adb_server has been set

ManubARM, 2020-02-20:
Hello,
One of our CI scripts is currently failing when trying to check if an Android device is connected or not. Referring to this file:
https://github.com/ARM-software/devlib/blob/master/devlib/utils/android.py

And considering the following functions:
```
 def __init__(self, device=None, timeout=None, platform=None, adb_server=None,
                 adb_as_root=False):
def adb_connect(device, timeout=None, attempts=MAX_ATTEMPTS):
def _ping(device):
```
By debugging I discovered that even when the the adb_server is set in the __init__(), the  adb_connect() and _ping() function do not "know" the server and try to connect locally.

Is there any particular reason for this?

marcbonnici, 2020-02-20:
It looks like these functions may be a missing that parameter.
Just to double check your issue, your environment is you have a remote adb server which should be used to connect to the device however it is instead falling back to a local server instance which is therefore failing to connect to the desired device?

On the system if you manually connect to the remote adb server beforehand does this function correctly?

ManubARM, 2020-02-20:
Yes that is our situation, here is a log:
[check_assets.log](https://github.com/ARM-software/devlib/files/4230648/check_assets.log)

You can find the problem by searching for:
`DEBUG:android:adb -s S2D7N19618010211 shell "ls /data/local/tmp > /dev/null"`

> On the system if you manually connect to the remote adb server beforehand does this function correctly?
If I run the script within the remote server it works. Unfortunately our CI system picks the working agents randomly, and need to use the remote agent with all the phones connected to

marcbonnici, 2020-02-20:
Thanks for the information. I've submitted a PR to ensure that we're using the supplied  `adb_server` in the adb helper functions:  
https://github.com/ARM-software/devlib/pull/469

Would you be able to pull this and see if it resolves your issue?

ManubARM, 2020-02-20:
Yes it resolves my issue
Thanks!

closed 2020-02-20 by ManubARM