ARM-software/devlib #498: TypeError: write() argument must be str, not bytes, in SshConnection

ManubARM, 2020-07-03:
Hello,
We are updating devlib in our CI system and we are getting this error when trying to run something in background:

```
[2020-07-02T15:10:23.959Z] Traceback (most recent call last):
[2020-07-02T15:10:23.959Z]   File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
[2020-07-02T15:10:23.959Z]     self.run()
[2020-07-02T15:10:23.959Z]   File "/usr/lib/python3.5/threading.py", line 862, in run
[2020-07-02T15:10:23.959Z]     self._target(*self._args, **self._kwargs)
[2020-07-02T15:10:23.959Z]   File "/home/bsgcomp/workspace/Tools_dev/lib/python3.5/site-packages/devlib/utils/ssh.py", line 658, in redirect_thread_f
[2020-07-02T15:10:23.959Z]     _read_paramiko_streams(stdout_in, stderr_in, select_timeout, callback, copy.copy(out_streams))
[2020-07-02T15:10:23.959Z]   File "/home/bsgcomp/workspace/Tools_dev/lib/python3.5/site-packages/devlib/utils/ssh.py", line 93, in _read_paramiko_streams
[2020-07-02T15:10:23.959Z]     return _read_paramiko_streams_internal(stdout, stderr, select_timeout, callback, init, chunk_size)
[2020-07-02T15:10:23.959Z]   File "/home/bsgcomp/workspace/Tools_dev/lib/python3.5/site-packages/devlib/utils/ssh.py", line 153, in _read_paramiko_streams_internal
[2020-07-02T15:10:23.959Z]     raise e
[2020-07-02T15:10:23.959Z]   File "/home/bsgcomp/workspace/Tools_dev/lib/python3.5/site-packages/devlib/utils/ssh.py", line 142, in _read_paramiko_streams_internal
[2020-07-02T15:10:23.959Z]     raise callback_excep
[2020-07-02T15:10:23.959Z]   File "/home/bsgcomp/workspace/Tools_dev/lib/python3.5/site-packages/devlib/utils/ssh.py", line 118, in read_channel
[2020-07-02T15:10:23.959Z]     callback_state = callback(callback_state, name, chunk)
[2020-07-02T15:10:23.959Z]   File "/home/bsgcomp/workspace/Tools_dev/lib/python3.5/site-packages/devlib/utils/ssh.py", line 645, in callback
[2020-07-02T15:10:23.959Z]     w.write(chunk)
[2020-07-02T15:10:23.959Z] TypeError: write() argument must be str, not bytes
```
I think it occurs only using python3.

Best,
Manuel

marcbonnici, 2020-07-03:
Hi, Would you be able to provide some additional context for how to trigger the bug? 
For example a log file or a set of commands leading up to the error.

ManubARM, 2020-07-03:
Unfortunately I don't have logs. But I can give more information about the code. 
We try to setup the test by starting a monitor command in the background like this:

```
    def monitor_frequency(self, prefix, working_dir, output_dir, as_root):
        logging.info("Starting monitor_frequency. dir: %s with frequency: %s" % (working_dir, self.mali_dvfs_table[0]))
        # Create a background connection and execute the script. Set a timeout of 5h
        command = "cd %s; timeout %s ./frequency_monitor.sh %s %s" % (working_dir, "18000", self.mali_dvfs_table[0], self.mali_path)

        out_log = open(output_dir + "/" + prefix + "_frequency.log", "w")
        err_log = open(output_dir + "/" + prefix + "_frequency.err", "w")
        return self.target.background(command, stdout=out_log, stderr=err_log, as_root=as_root)
```

even setting command = "ls" (for example) was enough to trigger the error

self.target should be a LinuxTarget

marcbonnici, 2020-07-03:
Thanks for the information. Could you try opening your `out_log` and `err_log` in binary mode `'wb'` instead?

ManubARM, 2020-07-03:
Yep, that fixed it!
Thanks

closed 2020-07-03 by ManubARM

ManubARM, 2020-07-03:
I am experiencing another problem. The pid returned from the background() seems to be something else.
The "timeout frequency_monitor.sh" command above has a while true loop in it. If I try to call the cancel() method it triggers an exception because the pid doesn't exist. And if I do a "ps -e" on the board, I can still see a "timeout" command going

marcbonnici, 2020-07-06:
The pid returned by the background object will be the shell that the background command was invoked in. If you wish to kill the background command can you try invoking the `cancel` method of the background object instead?

ManubARM, 2020-07-06:
I am using cancel method. So I am monitoring the board while the job is going

Here is the "ps -e" from the board before the kill:
```
 9320 pts/0    00:00:00 bash
 9342 ?        00:00:00 kworker/6:1
 9406 ?        00:00:00 sh
 9408 ?        00:00:00 sh
 9410 ?        00:00:00 timeout
 9411 ?        00:00:00 sh
 9418 ?        00:00:00 sshd
 9424 pts/1    00:00:00 bash
 9489 ?        00:00:00 sshd
 9493 pts/2    00:00:00 bash
 9556 ?        00:00:00 sshd
 9561 pts/3    00:00:00 bash
```

and this is the "ps-e" after the kill:
```
 9320 pts/0    00:00:00 bash
 9410 ?        00:00:00 timeout
 9411 ?        00:00:01 sh
 9418 ?        00:00:00 sshd
 9424 pts/1    00:00:00 bash
 9489 ?        00:00:00 sshd
 9493 pts/2    00:00:00 bash
 9556 ?        00:00:00 sshd
 9561 pts/3    00:00:00 bash
```

That's the error I get:
```
[2020-07-06T12:13:52.699Z] :ERROR:signal:Got exit code 1
[2020-07-06T12:13:52.699Z] :ERROR:signal:from: kill -SIGKILL -9406
[2020-07-06T12:13:52.699Z] :ERROR:signal:OUTPUT: bash: line 0: kill: (-9406) - No such process
[2020-07-06T12:13:52.699Z] :ERROR:signal:
```

The timeout is still there

Here is the code:
```
def setup(self, context):
     ...   
     self.background_pid = self.target.monitor_frequency(self.report_prefix, self.working_directory, context.output_directory)
    ...
```

```
def teardown(self, context):
        if self.background_pid:
            self.background_pid.cancel()
            self.background_pid = None
```

In fact that pid is already gone when the cancel is called

marcbonnici, 2020-07-06:
Can I check which version of devlib you are using? If you're not using the latest version from github could I ask you to try installing that? 

In the latest version of devlib the returned object from the `background` command should be a background object where the `cancel` method shouldn't try to kill a process that doesn't exist.

ManubARM, 2020-07-06:
I was using this one: devlib 1.2.1.dev3+af8c4715
I just updated and now I have devlib-1.2.1.dev3+9a2c4133
I will try now.

But I think for me the problem is that the pid is not the timeout and I cannot stop it.

marcbonnici, 2020-07-06:
Ok you're still on an up to date version so this functionality will not have changed.

I'm trying to recreate the your situation on my end, if I use the following code, the pid printed is of the `cd` command rather than the `sleep` command (however on my system the `cancel` method still appears to kill the `sleep` process as expected). So instead of running 2 commands in your background command could you just it as one single command?

```
bg_process = target.background('cd; sleep 120')
print(bg_process.pid)
sleep(5)
bg_process.cancel()
```

ManubARM, 2020-07-07:
I think I figure it out! It looks like the "timeout" is creating the problem. If I remove it, the command is correctly cancelled