ARM-software/devlib #549: perf record with Target.execute()

GavinTian1991, 2021-05-17:
Hi all,
        I am totally new to devlib. For now, I have already created Linux Target and made connection with the target machine. I can do some simple commands(zip, unzip, copy, etc.) by using `Target.execute()`.
        What I want to do now is to use command `perf record` for 10s (`perf record -g -F 99 -p $pid sleep 10`) and a `perf.data` file will be generated. After finishing perf record, I want to use `Target.pull()` to transfer perf.data file from the target device to the host machine. 
        However, based on the documentation, `Target.execute()` will return the output which will return all the `perf.data` contents to the host, no perf.data file will be generated on the target device, and it cost much more time to finish (10s recording but need over 90s to finish because of data return to the host). I also tried to set a timeout to `Target.execute()` and by using `Target.background()` to execute the perf record command and seems not working.
       I want to generate perf.data on the target machine and pull back to the host, any helps?
       Thanks in advance!

marcbonnici, 2021-05-18:
Hi,
`Target.execute` will return any output your command prints to the console, so if the command you execute expects to write to a file, the contents will not be returned automatically. I think `perf` should by default output directly to a `perf.data` file however you can customize this with the `-o` flag for perf, and you should be able to pull the output file as you planned once your command has completed. 

However depending on your usecase I might suggest instead of creating your own commands from scratch you may benefit from using the builtin devlib perf collector [1], which will take care of generating the commands based on your configuration and allow you to interact with it via our collector API [2] e.g. starting and stopping the collector, as well as pulling the data from the device.
Also depending on your desired usecase, we have a further abstraction in the form of an automation framework called "Workload Automation" [3] which uses devlib as a method of interacting with the device to automate your test cases which also has support for collecting perf data [4] while running a workload, which may be of interest to you.

Please let me know if there is any area you would like more information on.

[1] https://github.com/ARM-software/devlib/blob/b9374d530e610656f7b2350267afdcc6f43e65b5/devlib/collector/perf.py#L47
[2] https://devlib.readthedocs.io/en/latest/collectors.html#example
[3] https://workload-automation.readthedocs.io/en/latest/index.html
[4] https://workload-automation.readthedocs.io/en/latest/plugins/instruments.html#perf

GavinTian1991, 2021-05-19:
> Hi,
> `Target.execute` will return any output your command prints to the console, so if the command you execute expects to write to a file, the contents will not be returned automatically. I think [...]

Hi marc,
             Thank you for the instant reply! really appreciate!
              I tried the `PerfCollector ` today and seems to work perfectly!  One little question about the target API. The current git version is using `Target.background()` to execute the perf record. The devlib version I am using is based on the `Target.kick_off()`, any differences when doing with the perf command?
              Thank you again! I will also dig into the "Workload Automation" you recommend.

marcbonnici, 2021-05-19:
No problem, I'm glad to hear.

IIRC the switch between `background` and `kick_off` was part of a improvement in how we terminate the processes we'd created. It just changes the behind the scenes mechanism the command is executed on the target so there shouldn't be any functional difference in usage of perf. I think those changes should have made it to the latest release of devlib v1.3.1 so you should be to get the changes by updating your installed version if you wish.

GavinTian1991, 2021-05-19:
really helpful, thank you again!

closed 2021-05-19 by GavinTian1991