ARM-software/devlib #551: helped wanted: input password after Target.execute()

GavinTian1991, 2021-06-24:
Establishing ssh port mapping by using `Target.execute(ssh...)` on target machine, it needs to input password after execution of ssh port mapping command. How can I achieve it by using Target related command? Thanks a lot!

setrofim, 2021-06-24:
You can use `Target.background()` to get back a [Popen object](https://docs.python.org/3/library/subprocess.html#popen-objects) which would allow you to send the password to `STDIN`. However, if you can install [sshpass](https://linux.die.net/man/1/sshpass) on the target, that would provide a more reliable method.

GavinTian1991, 2021-06-25:
Hi, thank you for your instant answer. The `sshpass` command can do the work! Still, I want to achieve interactive way between devices which is very important to my further work.
I have read the doc, source code and debug by using `Target.background()`. The thing is that I used `LinuxTarget()` to create target and it will return a `ParamikoBackgroundCommand` object which has no attribute 'popen'. Below is the print out of object attributes:

```
{'chan': <paramiko.Channel 19 (open) window=2097152 -> <paramiko.Transport at 0x93459c88 (cipher aes128-ctr, 128 bits) (active; 1 open channel(s))>>,   
'as_root': True,   
'conn': <devlib.utils.ssh.SshConnection object at 0xffff93459b38>,   
'_pid': 9352, 
'_stdin': <paramiko.ChannelFile from <paramiko.Channel 19 (open) window=2097152 -> <paramiko.Transport at 0x93459c88 (cipher aes128-ctr, 128 bits) (active; 1 open channel(s))>>>, 
'_stdout': <_io.BufferedReader name=8>, 
'_stderr': <_io.BufferedReader name=10>, 
'redirect_thread': <Thread(Thread-13, started daemon 281472976941536)>}
```

I do notice that by using `LocalLinuxTarget()` to create target and target.backgroud() will return a `PopenBackgroundCommand` which has the `popen` attributes and I can do this:
```
t = LocalLinuxTarget()
p = t.background('echo $PWD')
output, error = p.popen.communicate()
```
However, it will run the command on host, not device.
The situation is that we have 1 host and 2 devices(d1 and d2), the 2 devices are connect with each other. The host can connect with d1 directly but cannot connect with d2. The ssh port mapping happen between d1 and d2 by creating devlib target on host and do remote command on d1.
Any ideas? Thanks in advance!

setrofim, 2021-06-25:
Ah, in that case, you should be write the password to `ParamikoBackgroundCommand.stdin`.

GavinTian1991, 2021-06-25:
Hi, thank you for your reply! The return `ParamikoBackgroundCommand.stdin` object is a `paramiko.channel.ChannelStdinFile` and below is the print out:
```
{'channel': <paramiko.Channel 19 (open) window=2097152 -> <paramiko.Transport at 0x93d69cf8 (cipher aes128-ctr, 128 bits) (active; 1 open channel(s))>>, 
'newlines': None, 
'_flags': 1, 
'_bufsize': 8192, 
'_wbuffer': <_io.BytesIO object at 0xffff93d38728>, 
'_rbuffer': b'', 
'_at_trailing_cr': False, 
'_closed': False, 
'_pos': 0, 
'_realpos': 0, 
'_size': 0}
```
The thing is that when I tried to input password and it said:
```
File "/usr/local/lib/python3.7/site-packages/paramiko/file.py", line 403, in write
    raise IOError("File not open for writing")
```
I also checked the code in paramiko and by using `stdin.writable()` , the output is False. The original value of `_flags` is 0 but it turned to 1 which can be seen above. Also, the `_wbuffer` is an empty {}. 
I am wondering is it maybe caused by the device unique or the ssh command I used here is not quite fit for the `Target.background()` situation?
The ssh command I am using here is like:
```
ssh -g -N -f -L port:ip_addr1 root@ip_addr2
Password:
```
The command require me to input password manually.
Thanks in advance!

marcbonnici, 2021-06-29:
Hi, 
> I also checked the code in paramiko and by using stdin.writable() , the output is False. 

Thanks for investigating this, it looks like we had a bug causing our stdin stream to be opened as read only. I've created a PR to fix this: https://github.com/ARM-software/devlib/pull/552

However as a potentially easier way to accomplish your goal are you able to use key based authentication instead of a passwords? Either adding the keys of d1 to d2 or by pushing the required key for d2 to d1 from the host e.g.
```
target.push(keyfile, remote_keyfile_path)
cmd = 'ssh -i {} -g -N -f -L port:ip_addr1:22 root@ip_addr2'
target.background(cmd.format(remote_keyfile_path))
```

closed 2021-09-04 by GavinTian1991