ARM-software/devlib #614: bug: Target parameter, max_sync, isn't used when calling connect outside __init__

benhor01, 2023-01-31:
The Target parameter, max_sync, isn't used when calling connect outside` __init__`.
I found this while trying to connect to the total compute FVP with LISA.
This could be fixed by doing something along the lines of:
```
--- a/external/devlib/devlib/target.py
+++ b/external/devlib/devlib/target.py
@@ -338,6 +338,7 @@ class Target(object):
         self.shell_prompt = bytes_regex(shell_prompt)
         self.conn_cls = conn_cls
         self.is_container = is_container
+        self.max_async = max_async
         self.logger = logging.getLogger(self.__class__.__name__)
         self._installed_binaries = {}
         self._installed_modules = {}
@@ -387,7 +388,9 @@ class Target(object):
     # connection and initialization
 
     @asyn.asyncf
-    async def connect(self, timeout=None, check_boot_completed=True, max_async=50):
+    async def connect(self, timeout=None, check_boot_completed=True, max_async=None):
+        if max_async is None:
+            max_async = self.max_async
         self.platform.init_target_connection(self)
         # Forcefully set the thread-local value for the connection, with the
         # timeout we want
```

marcbonnici, 2023-02-01:
Thanks for spotting that. I've create a PR #615 to address the issue, please let me know if this resolves the problem for you.

benhor01, 2023-02-02:
Yes, thanks for the speedy fix. Along with https://github.com/ARM-software/lisa/pull/1936/commits/93e8f27043cb76ca35531872f614ebcadc472082
this fixes my problem with using max_async with lisa.

closed 2023-02-03 by marcbonnici