ARM-software/devlib #693: Invalid PID wheb trying to kill BackgroundCommand

ep1cman, 2024-06-14:
I have started a background task as follows:

```
self._process = self._target.background(f"poller -l {','.join([str(x) for x in self._files.keys()])} {' '.join(self._files.values())}", as_root=self._as_root)
```

When I then try to run:

```
self._process.cancel()
```
I see the following error:

```
Command '/home/appa/devlib-target/bin/busybox kill -15 -329893' returned non-zero exit status 1.
OUTPUT: kill: can't kill pid -329893: No such process
```

Checking on the device I can see that the PID does exist:
```
root      329893  0.0  0.0   2060   500 ?        S    10:44   0:00 sh -c sh -c 'export PATH=/home/appa/devlib-target/bin:$PATH && export LC_ALL=C && poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq'
root      329894  0.0  0.0   2060   472 ?        S    10:44   0:00 sh -c export PATH=/home/appa/devlib-target/bin:$PATH && export LC_ALL=C && poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
root      329895  0.0  0.0    936     4 ?        S    10:44   0:00 poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
appa      330021  0.0  0.0   8576   648 pts/0    S+   10:45   0:00 grep --color=auto poller
```

It seems the issue is the preceding `-`, if I run the command manually without it (`/home/appa/devlib-target/bin/busybox kill -15 329893`) the process is killed as expected

ep1cman, 2024-06-14:
It seems that when it is being wrapped by sudo, the PID returned is not the PGID.

```
$ ps -efj | grep 373747
UID          PID    PPID    PGID     SID  C STIME TTY          TIME CMD
root      373747  370822  373747  373747  0 11:43 ?        00:00:00 sudo -k -p   -S -- sh -c printf '%s ' $$; exec sh -c 'sh -c '"'"'export PATH=/home/appa/devlib-target/bin:$PATH && export LC_ALL=C && poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq'"'"'  '
root      373748  373747  373747  373747  0 11:43 ?        00:00:00 sh -c sh -c 'export PATH=/home/appa/devlib-target/bin:$PATH && export LC_ALL=C && poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq'
root      373749  373748  373747  373747  0 11:43 ?        00:00:00 sh -c export PATH=/home/appa/devlib-target/bin:$PATH && export LC_ALL=C && poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
root      373750  373749  373747  373747  0 11:43 ?        00:00:00 poller -l 0 /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq
appa      373769  352164  373768  352164  0 11:43 pts/0    00:00:00 grep --color=auto 373747
```

The PID sent to the kill command in this case was `373748`, but you can see the PGID is actually `373747` which is the the outer sudo call

douglas-raillard-arm, 2024-07-26:
Interesting, thanks for reporting.

In case someone has the same idea, this PR https://github.com/ARM-software/devlib/pull/689 does not provide a workaround, since SIGTERM, SIGQUIT and SIGKILL are still handled in the same way as currently.